Napsty/check_smart #16: ATA read errors

deric, 2015-08-14:
I have 2 ATA disks, this one looks ok:
```
OK: no SMART errors detected. |Raw_Read_Error_Rate=0 Throughput_Performance=82 Spin_Up_Time=0 Start_Stop_Count=4 Reallocated_Sector_Ct=0 Seek_Error_Rate=0 Seek_Time_Performance=27 Power_On_Hours=5690 Spin_Retry_Count=0 Power_Cycle_Count=4 Power-Off_Retract_Count=4 Load_Cycle_Count=4 Temperature_Celsius=34 Reallocated_Event_Count=0 Current_Pending_Sector=0 Offline_Uncorrectable=0 UDMA_CRC_Error_Count=0
```
and the other one has pretty high `Raw_Read_Error_Rate`, though no warning is shown.
```
OK: no SMART errors detected. |Raw_Read_Error_Rate=3080195 Throughput_Performance=78 Spin_Up_Time=0 Start_Stop_Count=4 Reallocated_Sector_Ct=0 Seek_Error_Rate=0 Seek_Time_Performance=26 Power_On_Hours=5690 Spin_Retry_Count=0 Power_Cycle_Count=4 Power-Off_Retract_Count=4 Load_Cycle_Count=4 Temperature_Celsius=37 Reallocated_Event_Count=0 Current_Pending_Sector=0 Offline_Uncorrectable=0 UDMA_CRC_Error_Count=0
```
`smartctl -a` show some errors as well:
```
Error 1 occurred at disk power-on lifetime: 4281 hours (178 days + 9 hours)
  When the command that caused the error occurred, the device was active or idle.

  After command completion occurred, registers were:
  ER ST SC SN CL CH DH
  -- -- -- -- -- -- --
  40 51 10 f8 9e 19 00  Error: WP at LBA = 0x00199ef8 = 1679096

  Commands leading to the command that caused the error were:
  CR FR SC SN CL CH DH DC   Powered_Up_Time  Command/Feature_Name
  -- -- -- -- -- -- -- --  ----------------  --------------------
  61 b0 c0 78 10 f1 40 00   3d+18:58:53.439  WRITE FPDMA QUEUED
  60 10 b8 f8 9e 19 40 00   3d+18:58:53.414  READ FPDMA QUEUED
  60 10 b0 70 87 19 40 00   3d+18:58:53.414  READ FPDMA QUEUED
  60 10 a8 c8 86 19 40 00   3d+18:58:53.409  READ FPDMA QUEUED
  ea 00 00 00 00 00 a0 00   3d+18:58:53.386  FLUSH CACHE EXT
```
is it possible to detect this type of errors? Whole output is here http://paste.ubuntu.com/12078592/

Napsty, 2018-04-28:
I just had a case where check_smart.pl detected pending sectors (which is good) however the results of a self-test clearly shows that there are read failures. However the value of Raw_Read_Error_Rate is 0 so this is not a good indicator:

```
SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000b   100   100   016    Pre-fail  Always       -       0
  2 Throughput_Performance  0x0005   141   141   054    Pre-fail  Offline      -       72
  3 Spin_Up_Time            0x0007   127   127   024    Pre-fail  Always       -       180 (Average 180)
  4 Start_Stop_Count        0x0012   100   100   000    Old_age   Always       -       29
  5 Reallocated_Sector_Ct   0x0033   100   100   005    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x000b   100   100   067    Pre-fail  Always       -       0
  8 Seek_Time_Performance   0x0005   113   113   020    Pre-fail  Offline      -       35
  9 Power_On_Hours          0x0012   099   099   000    Old_age   Always       -       11032
 10 Spin_Retry_Count        0x0013   100   100   060    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       29
192 Power-Off_Retract_Count 0x0032   100   100   000    Old_age   Always       -       77
193 Load_Cycle_Count        0x0012   100   100   000    Old_age   Always       -       77
194 Temperature_Celsius     0x0002   193   193   000    Old_age   Always       -       31 (Min/Max 22/35)
196 Reallocated_Event_Count 0x0032   100   100   000    Old_age   Always       -       0
197 Current_Pending_Sector  0x0022   100   100   000    Old_age   Always       -       16
198 Offline_Uncorrectable   0x0008   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x000a   200   200   000    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Extended offline    Completed: read failure       10%     10926         962363536

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.
```

But in general you're right. We should be able to see the read errors from the output and alert.

Napsty, 2018-04-28:
By the way, to my current knowledge errors like these:

Error 1 occurred at disk power-on lifetime: 4281 hours (178 days + 9 hours)
  When the command that caused the error occurred, the device was active or idle.

May also occur when there was a sudden power-outage. You can just pull the power plug(s) of your computer/server. The disks won't like it ;-)
On the other hand this doesn't forcibly mean that the disk is now failing.

Napsty, 2018-04-29:
I just created the following PR #25 which introduces a new parameter (-s) to use the selftest log as additional check. 

In your case, @deric , I recommend to run an extended self test of the affected drive you believe could have errors (e.g. `smartctl -t long /dev/sda`) and then run check_smart.pl with the -s parameter e.g. `./check_smart.pl -i ata -d /dev/sda -s`.

Note that as of today it's not fully merged yet. This feature will be released in version 5.10.

closed 2018-04-29 by Napsty