Napsty/check_smart #26: Reallocated_Sector_Ct & Seek_Error_Rate

brentclark, 2018-06-04:
Good day

Can your script please too monitor Reallocated_Sector_Ct & Seek_Error_Rate?

Kind Regards
Brent Clark

Rohlik, 2019-04-01:
This would be great, because this is my S.M.A.R.T. output and current version tell me that everything is OK.
```
 SMART Attributes Data Structure revision number: 16
 Vendor Specific SMART Attributes with Thresholds:
 ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
   1 Raw_Read_Error_Rate     0x002f   200   200   051    Pre-fail  Always       -       13
   3 Spin_Up_Time            0x0027   193   193   021    Pre-fail  Always       -       3308
   4 Start_Stop_Count        0x0032   100   100   000    Old_age   Always       -       15
   5 Reallocated_Sector_Ct   0x0033   200   200   140    Pre-fail  Always       -       1
   7 Seek_Error_Rate         0x002e   200   200   000    Old_age   Always       -       0
   9 Power_On_Hours          0x0032   057   057   000    Old_age   Always       -       31762
  10 Spin_Retry_Count        0x0032   100   253   000    Old_age   Always       -       0
  11 Calibration_Retry_Count 0x0032   100   253   000    Old_age   Always       -       0
  12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       7
 192 Power-Off_Retract_Count 0x0032   200   200   000    Old_age   Always       -       4
 193 Load_Cycle_Count        0x0032   200   200   000    Old_age   Always       -       77
 194 Temperature_Celsius     0x0022   123   113   000    Old_age   Always       -       27
 196 Reallocated_Event_Count 0x0032   199   199   000    Old_age   Always       -       1
 197 Current_Pending_Sector  0x0032   200   200   000    Old_age   Always       -       0
 198 Offline_Uncorrectable   0x0030   200   200   000    Old_age   Offline      -       0
 199 UDMA_CRC_Error_Count    0x0032   200   200   000    Old_age   Always       -       0
 200 Multi_Zone_Error_Rate   0x0008   200   200   000    Old_age   Offline      -       0
```

Napsty, 2019-04-01:
Yes, definitely on my todo list. 
I think it'll be something like the current "-b" parameter to increase the thresholds if needed. I'm actually thinking of deprecating the "-b" parameter and replace it with with the well known "-w" and "-c" parameters but use different thresholds for each given attribute. Not sure yet how to implement. 
Open for suggestions.

Napsty, 2019-06-04:
Here's an additional SMART attribute list of a SSD which seems to be dying. 

``` 
SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0033   099   099   ---    Pre-fail  Always       -       16
  9 Power_On_Hours          0x0032   095   095   ---    Old_age   Always       -       25045
 12 Power_Cycle_Count       0x0032   099   099   ---    Old_age   Always       -       890
175 Program_Fail_Count_Chip 0x0032   099   099   ---    Old_age   Always       -       11
176 Erase_Fail_Count_Chip   0x0032   100   100   ---    Old_age   Always       -       0
177 Wear_Leveling_Count     0x0013   076   076   ---    Pre-fail  Always       -       867
178 Used_Rsvd_Blk_Cnt_Chip  0x0013   080   080   ---    Pre-fail  Always       -       396
179 Used_Rsvd_Blk_Cnt_Tot   0x0013   082   082   ---    Pre-fail  Always       -       722
180 Unused_Rsvd_Blk_Cnt_Tot 0x0013   082   082   ---    Pre-fail  Always       -       3310
181 Program_Fail_Cnt_Total  0x0032   099   099   ---    Old_age   Always       -       16
182 Erase_Fail_Count_Total  0x0032   100   100   ---    Old_age   Always       -       0
183 Runtime_Bad_Block       0x0013   099   099   ---    Pre-fail  Always       -       16
187 Uncorrectable_Error_Cnt 0x0032   067   067   ---    Old_age   Always       -       33281
195 ECC_Error_Rate          0x001a   001   001   ---    Old_age   Always       -       33281
198 Offline_Uncorrectable   0x0030   100   100   ---    Old_age   Offline      -       0
199 CRC_Error_Count         0x003e   253   253   ---    Old_age   Always       -       0
232 Available_Reservd_Space 0x0013   080   080   ---    Pre-fail  Always       -       1620
241 Total_LBAs_Written      0x0032   002   002   ---    Old_age   Always       -       4237932280
242 Total_LBAs_Read         0x0032   035   035   ---    Old_age   Always       -       2781507796
```

Here `Reallocated_Sector_Ct` is a good indicator, too. 
Other attributes which look promising to identify a failing drive: `Uncorrectable_Error_Cnt` , `ECC_Error_Rate`, `Offline_Uncorrectable`, `CRC_Error_Count`, `Program_Fail_Cnt_Total`, `Runtime_Bad_Block`. Some of these attributes apply only to SSD's, some to all drives. Depending on the drive vendor and model, some attributes may not show up. 

However I would not use `Seek_Error_Rate` as a failing indicator. This value can sometimes spike up and become 0 again. Seen in the past and shown here in this graph:

[image]

So basically what I am working on is another list (raw_check_list) of attributes similar to the exclude_list (ignore list). There will be a default list of attributes to be checked for their raw values. This default list can be overwritten with a paramter `-r`.

Napsty, 2019-06-04:
Released: https://www.claudiokuenzler.com/blog/864/major-release-check_smart-multi-attribute-value-checks-warning-thresholds

closed 2019-06-04 by Napsty