Napsty/check_smart #30: Make -g option work for 3ware interface

karaluh, 2019-01-08:
This is already supported for megaraid interface.

Napsty, 2019-11-22:
Hi @karaluh 
Can you try it with this version and let me know if this works?
https://github.com/Napsty/check_smart/blob/issue-30/check_smart.pl 

I don't have any 3ware raid controllers at hand so I can't test it properly.

Napsty, 2019-11-22:
Hi @karaluh 
I was able to add cciss, too. As I have HP serves to test with, this seems to work correctly. 
At the same time I also adjusted the output of the plugin when `-g` is used in combination with megaraid, cciss or 3ware glob.

Before: 

```
# ./check_smart.pl -g /dev/sda -i 'cciss,[1-3]'
OK: [/dev/sda] - Device is clean --- [/dev/sda] - Device is clean --- [/dev/sda] - Device is clean|
```

New:

```
# ./check_smart.pl -g /dev/sda -i 'cciss,[1-3]'
OK: [cciss,1] - Device is clean --- [cciss,2] - Device is clean --- [cciss,3] - Device is clean|
```

Napsty, 2019-11-29:
Solved in Version 6.6.

closed 2019-11-29 by Napsty