Napsty/check_smart #33: Is it possible to hide results like "No health status line found"

HHawk, 2019-05-21:
Hi there,

I am running this script to check all available harddisks for smart errors. I got this working, however it showing multiple "No health status line found" results within the Nagios GUI.

Is there a way to hide this from the output?

Here is a result of what the outputs looks like now within the Nagios GUI:
`UNKNOWN: [/dev/sda] - Device is clean --- [/dev/sda] - Device is clean --- [/dev/sda] - Device is clean --- [/dev/sda] - Device is clean --- [/dev/sda] - No health status line found --- [/dev/sda] - No health status line found --- [/dev/sda] - No health status line found --- [/dev/sda] - No health status line found --- [/dev/sda] - No health status line found --- [/dev/sda] - No health status line found `

Would be great to hide those "No health status line" results.

HHawk, 2019-05-21:
By the way, a small follow-up; I noticed all my drives are being displayed as /dev/sda. I understand this is caused by the way I use the plugin. For example:

`command[check_smart_all]=sudo /usr/lib64/nagios/plugins/check_smart.pl -g /dev/sda -i "megaraid,[0-8]"`

The reason for this, is that I only have /dev/sda, but no other drives being displayed like that?
For example from the command line, how I normally check the disks behind RAID is like this:

```
smartctl -a -d megaraid,1 /dev/sda               #For the 1st HDD
smartctl -a -d megaraid,2 /dev/sda               #For the 2nd HDD
smartctl -a -d megaraid,3 /dev/sda               #For the 3d HDD
etc.
```

If I use "smartctl -a -d megaraid,1 /dev/sdb" it will output: **Smartctl open device: /dev/sdb [megaraid_disk_00] failed: No such device**

Is this normal?

Regards

Napsty, 2019-05-22:
> By the way, a small follow-up; I noticed all my drives are being displayed as /dev/sda. I understand this is caused by the way I use the plugin.

You're using a hardware raid controller. Each raid array is presented as a drive to the OS. Hence several drives can be behind /dev/sda. That's not the plugin. 

Also using the -g parameter for hardware raids is not recommended because the positions in the raid array (whether this is cciss or megaraid or something else) might change after a reboot. So you should consider doing single drive checks. Example for your 1st HDD:

```
./check_smart.pl -d /dev/sda -i megaraid,1
```

> Update: I noticed another minor issue. It also checks the "backplane" as healthy.

Again, that's how the OS implements the raid controller. One of the megaraid internal positions (usually position 0 but this may differ) is always the controller itself.

closed 2019-05-22 by Napsty