Napsty/check_smart #39: -w warning option never results in a warning

joopmartens, 2019-07-14:
The warning option never seems to result in an actual warning result.
In the below example the Temperature_Celsius and Power-Off_Retract_Count do have higher values than the supplied warning thresholds but the result is still: OK: no SMART errors detected.

# ./check_smart.pl -d /dev/sda -i auto -w 'Temperature_Celsius=40,Power-Off_Retract_Count=45' --debug
Found /dev/sda
###########################################################
(debug) CHECK 1: getting overall SMART health status for /dev/sda
###########################################################

(debug) executing:
sudo smartctl -d auto -H /dev/sda

(debug) output:
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-18-pve] (local build)
 Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

 === START OF READ SMART DATA SECTION ===
 SMART overall-health self-assessment test result: PASSED

(debug) parsing line:
SMART overall-health self-assessment test result: PASSED

(debug) found string 'PASSED'; status OK

###########################################################
(debug) CHECK 2: getting silent SMART health check
###########################################################

(debug) executing:
sudo smartctl -d auto -q silent -A /dev/sda

(debug) exit code:
0

(debug) zero exit code, status OK

###########################################################
(debug) CHECK 3: getting detailed statistics
(debug) information contains a few more potential trouble spots
(debug) plus, we can also use the information for perfdata/graphing
###########################################################

(debug) executing:
sudo smartctl -d auto -A /dev/sda

(debug) output:
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.18-18-pve] (local build)
 Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

 === START OF READ SMART DATA SECTION ===
 SMART Attributes Data Structure revision number: 16
 Vendor Specific SMART Attributes with Thresholds:
 ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
   1 Raw_Read_Error_Rate     0x002f   200   200   051    Pre-fail  Always       -       0
   3 Spin_Up_Time            0x0027   179   138   021    Pre-fail  Always       -       10008
   4 Start_Stop_Count        0x0032   100   100   000    Old_age   Always       -       85
   5 Reallocated_Sector_Ct   0x0033   200   200   140    Pre-fail  Always       -       0
   7 Seek_Error_Rate         0x002e   200   200   000    Old_age   Always       -       0
   9 Power_On_Hours          0x0032   068   068   000    Old_age   Always       -       23908
  10 Spin_Retry_Count        0x0032   100   253   000    Old_age   Always       -       0
  11 Calibration_Retry_Count 0x0032   100   253   000    Old_age   Always       -       0
  12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       82
 192 Power-Off_Retract_Count 0x0032   200   200   000    Old_age   Always       -       48
 193 Load_Cycle_Count        0x0032   200   200   000    Old_age   Always       -       37
 194 Temperature_Celsius     0x0022   105   091   000    Old_age   Always       -       47
 196 Reallocated_Event_Count 0x0032   200   200   000    Old_age   Always       -       0
 197 Current_Pending_Sector  0x0032   200   200   000    Old_age   Always       -       0
 198 Offline_Uncorrectable   0x0030   200   200   000    Old_age   Offline      -       0
 199 UDMA_CRC_Error_Count    0x0032   200   200   000    Old_age   Always       -       0
 200 Multi_Zone_Error_Rate   0x0008   200   200   000    Old_age   Offline      -       0

(debug) Raw Check List: Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block
(debug) Exclude List:
(debug) Warning Thresholds:
Power-Off_Retract_Count=45
Temperature_Celsius=40

(debug) Raw_Read_Error_Rate not in raw check list (raw value: 0)

(debug) Spin_Up_Time not in raw check list (raw value: 10008)

(debug) Start_Stop_Count not in raw check list (raw value: 85)

(debug) Reallocated_Sector_Ct is OK (0)

(debug) Seek_Error_Rate not in raw check list (raw value: 0)

(debug) Power_On_Hours not in raw check list (raw value: 23908)

(debug) Spin_Retry_Count not in raw check list (raw value: 0)

(debug) Calibration_Retry_Count not in raw check list (raw value: 0)

(debug) Power_Cycle_Count not in raw check list (raw value: 82)

(debug) Power-Off_Retract_Count not in raw check list (raw value: 48)

(debug) Load_Cycle_Count not in raw check list (raw value: 37)

(debug) Temperature_Celsius not in raw check list (raw value: 47)

(debug) Reallocated_Event_Count not in raw check list (raw value: 0)

(debug) Current_Pending_Sector is OK (0)

(debug) Offline_Uncorrectable is OK (0)

(debug) UDMA_CRC_Error_Count not in raw check list (raw value: 0)

(debug) Multi_Zone_Error_Rate not in raw check list (raw value: 0)

(debug) gathered perfdata:
Raw_Read_Error_Rate=0 Spin_Up_Time=10008 Start_Stop_Count=85 Reallocated_Sector_Ct=0 Seek_Error_Rate=0 Power_On_Hours=23908 Spin_Retry_Count=0 Calibration_Retry_Count=0 Power_Cycle_Count=82 Power-Off_Retract_Count=48 Load_Cycle_Count=37 Temperature_Celsius=47 Reallocated_Event_Count=0 Current_Pending_Sector=0 Offline_Uncorrectable=0 UDMA_CRC_Error_Count=0 Multi_Zone_Error_Rate=0

###########################################################
(debug) LOCAL STATUS: OK, FINAL STATUS: OK
###########################################################

(debug) final status/output: OK
OK: no SMART errors detected. |Raw_Read_Error_Rate=0 Spin_Up_Time=10008 Start_Stop_Count=85 Reallocated_Sector_Ct=0 Seek_Error_Rate=0 Power_On_Hours=23908 Spin_Retry_Count=0 Calibration_Retry_Count=0 Power_Cycle_Count=82 Power-Off_Retract_Count=48 Load_Cycle_Count=37 Temperature_Celsius=47 Reallocated_Event_Count=0 Current_Pending_Sector=0 Offline_Uncorrectable=0 UDMA_CRC_Error_Count=0 Multi_Zone_Error_Rate=0

Napsty, 2019-07-18:
Hello

It's because these attributes (Temperature_Celsius, Power-Off_Retract_Count=45) are not in the raw check list:

```
(debug) Temperature_Celsius not in raw check list (raw value: 47)
(debug) Power-Off_Retract_Count not in raw check list (raw value: 48)
```

If you want them checked by the plugin, you need to append the raw list using `-r`: `Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Temperature_Celsius,Power-Off_Retract_Count`.

joopmartens, 2019-07-21:
Indeed, by adding it to the raw list it is working just fine.
Thanks!

closed 2019-07-21 by joopmartens