Napsty/check_smart #52: check_smart does not support General Purpose Log

hansmi, 2020-05-12:
One of my computers has an SSD (model ADATA SP600NS34) which doesn't report its used life as a SMART attribute. It does, however, report it in the "General Purpose Log".

```
$ smartctl -l ssd /dev/sda
…
Device Statistics (GP Log 0x04)
Page  Offset Size        Value Flags Description
0x07  =====  =               =  ===  == Solid State Device Statistics (rev 1) ==
0x07  0x008  1              56  N--  Percentage Used Endurance Indicator
                                |||_ C monitored condition met
                                ||__ D supports DSN
                                |___ N normalized value
```

Supported tables can be listed with `smartctl -l devstat,0` and each supported table can be retrieved using `smartctl -l devstat,<page>`. `-l ssd` is equivalent to `-l devstat,7`. The SMART attributes for the aforementioned SSD:

```
=== START OF READ SMART DATA SECTION ===
SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000a   100   100   000    Old_age   Always       -       0
  2 Throughput_Performance  0x0005   100   100   050    Pre-fail  Offline      -       0
  3 Spin_Up_Time            0x0007   100   100   050    Pre-fail  Always       -       0
  5 Reallocated_Sector_Ct   0x0013   100   100   050    Pre-fail  Always       -       0
  7 Unknown_SSD_Attribute   0x000b   100   100   050    Pre-fail  Always       -       0
  8 Unknown_SSD_Attribute   0x0005   100   100   050    Pre-fail  Offline      -       0
  9 Power_On_Hours          0x0012   100   100   000    Old_age   Always       -       18904
 10 Unknown_SSD_Attribute   0x0013   100   100   050    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0012   100   100   000    Old_age   Always       -       658
167 Unknown_Attribute       0x0022   100   100   000    Old_age   Always       -       0
168 Unknown_Attribute       0x0012   100   100   000    Old_age   Always       -       93
169 Unknown_Attribute       0x0013   100   100   010    Pre-fail  Always       -       4295950346
170 Unknown_Attribute       0x0013   100   100   010    Pre-fail  Always       -       0
173 Unknown_Attribute       0x0012   143   143   000    Old_age   Always       -       25862866203
175 Program_Fail_Count_Chip 0x0013   100   100   010    Pre-fail  Always       -       0
180 Unused_Rsvd_Blk_Cnt_Tot 0x0033   099   099   020    Pre-fail  Always       -       1089
192 Power-Off_Retract_Count 0x0012   100   100   000    Old_age   Always       -       93
194 Temperature_Celsius     0x0022   063   063   030    Old_age   Always       -       37 (Min/Max 28/38)
197 Current_Pending_Sector  0x0012   100   100   000    Old_age   Always       -       0
231 Temperature_Celsius     0x0033   100   100   005    Pre-fail  Always       -       0
233 Media_Wearout_Indicator 0x0032   100   100   000    Old_age   Always       -       691419471104
234 Unknown_Attribute       0x0032   100   100   000    Old_age   Always       -       1075000130176
240 Unknown_SSD_Attribute   0x0013   100   100   050    Pre-fail  Always       -       0
241 Total_LBAs_Written      0x0032   100   100   000    Old_age   Always       -       32170085668
242 Total_LBAs_Read         0x0032   100   100   000    Old_age   Always       -       9610546032
```

`check_smart` could retrieve the general purpose log pages and treat them similar to the SMART attributes.

Napsty, 2020-05-12:
Interesting case. Just out of curiousity, what does `smartctl -a /dev/sda` show on that SSD? It might also be a missing device recognition in smartmontools' smartctl to show the attributes directly.

hansmi, 2020-05-13:
> what does `smartctl -a /dev/sda` show on that SSD?

I'm including the output below. The machine in question runs Debian Buster.

> ADATA SP600 devics should be recognizable by smartc[t]l: […]

The regular expression is for `ADATA SP600` (among others) while the device uses `ADATA SP600NS34`. The [structure description in drivedb.h](https://www.smartmontools.org/browser/trunk/smartmontools/drivedb.h?rev=5047#L42) says that the "regular expressions for drive model and firmware must match the full string". So this is definitely a problem, but not the cause: none of the attributes seems to be for the lifetime percentage.

```
smartctl 6.6 2017-11-05 r4594 [x86_64-linux-4.19.0-8-amd64] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Device Model:     ADATA SP600NS34
Serial Number:    XXXXXXXXXXXX
LU WWN Device Id: 5 000000 000000000
Firmware Version: 5.8f
User Capacity:    256'060'514'304 bytes [256 GB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
Device is:        Not in smartctl database [for details use: -P showall]
ATA Version is:   ACS-2 (minor revision not indicated)
SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Wed May 13 XX:XX:XX 2020 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00)	Offline data collection activity
					was never started.
					Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0)	The previous self-test routine completed
					without error or no self-test has ever 
					been run.
Total time to complete Offline 
data collection: 		(   32) seconds.
Offline data collection
capabilities: 			 (0x5b) SMART execute Offline immediate.
					Auto Offline data collection on/off support.
					Suspend Offline collection upon new
					command.
					Offline surface scan supported.
					Self-test supported.
					No Conveyance Self-test supported.
					Selective Self-test supported.
SMART capabilities:            (0x0003)	Saves SMART data before entering
					power-saving mode.
					Supports SMART auto save timer.
Error logging capability:        (0x01)	Error logging supported.
					General Purpose Logging supported.
Short self-test routine 
recommended polling time: 	 (   1) minutes.
Extended self-test routine
recommended polling time: 	 (   1) minutes.
SCT capabilities: 	       (0x0039)	SCT Status supported.
					SCT Error Recovery Control supported.
					SCT Feature Control supported.
					SCT Data Table supported.

SMART Attributes Data Structure revision number: 16
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000a   100   100   000    Old_age   Always       -       0
  2 Throughput_Performance  0x0005   100   100   050    Pre-fail  Offline      -       0
  3 Spin_Up_Time            0x0007   100   100   050    Pre-fail  Always       -       0
  5 Reallocated_Sector_Ct   0x0013   100   100   050    Pre-fail  Always       -       0
  7 Unknown_SSD_Attribute   0x000b   100   100   050    Pre-fail  Always       -       0
  8 Unknown_SSD_Attribute   0x0005   100   100   050    Pre-fail  Offline      -       0
  9 Power_On_Hours          0x0012   100   100   000    Old_age   Always       -       18904
 10 Unknown_SSD_Attribute   0x0013   100   100   050    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0012   100   100   000    Old_age   Always       -       659
167 Unknown_Attribute       0x0022   100   100   000    Old_age   Always       -       0
168 Unknown_Attribute       0x0012   100   100   000    Old_age   Always       -       93
169 Unknown_Attribute       0x0013   100   100   010    Pre-fail  Always       -       4295950346
170 Unknown_Attribute       0x0013   100   100   010    Pre-fail  Always       -       0
173 Unknown_Attribute       0x0012   143   143   000    Old_age   Always       -       25862866203
175 Program_Fail_Count_Chip 0x0013   100   100   010    Pre-fail  Always       -       0
180 Unused_Rsvd_Blk_Cnt_Tot 0x0033   099   099   020    Pre-fail  Always       -       1089
192 Power-Off_Retract_Count 0x0012   100   100   000    Old_age   Always       -       93
194 Temperature_Celsius     0x0022   064   064   030    Old_age   Always       -       36 (Min/Max 28/37)
197 Current_Pending_Sector  0x0012   100   100   000    Old_age   Always       -       0
231 Temperature_Celsius     0x0033   100   100   005    Pre-fail  Always       -       0
233 Media_Wearout_Indicator 0x0032   100   100   000    Old_age   Always       -       691451864320
234 Unknown_Attribute       0x0032   100   100   000    Old_age   Always       -       1075055336192
240 Unknown_SSD_Attribute   0x0013   100   100   050    Pre-fail  Always       -       0
241 Total_LBAs_Written      0x0032   100   100   000    Old_age   Always       -       32170644607
242 Total_LBAs_Read         0x0032   100   100   000    Old_age   Always       -       9612320576

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
No self-tests have been logged.  [To run self-tests, use: smartctl -t]

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.
```

Napsty, 2022-04-22:
@hansmi does it work since https://github.com/smartmontools/smartmontools/pull/53 ?

hansmi, 2022-04-25:
> @hansmi does it work since [smartmontools/smartmontools#53](https://github.com/smartmontools/smartmontools/pull/53) ?

The SSD was recognized after that, but as described in comment https://github.com/Napsty/check_smart/issues/52#issuecomment-628024722 the model in question doesn't report its lifetime as a plain SMART value. There is a value in the General Purpose Log.