Napsty/check_smart #57: -g flag doesn't work with nvme-devices

extrad, 2020-07-23:
Hi,

-g flag doesn't work with nvme-devices the way it works with sata/meragaid/cciss:

Device list:
`ls /dev/nvme*
/dev/nvme0  /dev/nvme0n1  /dev/nvme0n1p1  /dev/nvme1  /dev/nvme1n1  /dev/nvme1n1p1  /dev/nvme2  /dev/nvme2n1  /dev/nvme2n1p1  /dev/nvme3  /dev/nvme3n1  /dev/nvme3n1p1`

Plugin Run:
`./check_smart.pl -g /dev/nvme[0-3] -i auto
OK: [/dev/nvme0] - Device is clean|`

Expected output:
`OK: [nvme,0] - Device is clean --- [nvme,1] - Device is clean --- [nvme,2] - Device is clean --- [nvme,3] - Device is clean |`

Napsty, 2020-07-23:
Did you try it with `./check_smart.pl -g "/dev/nvme[0-3]" -i nvme` ?

extrad, 2020-07-23:
Sorry, my fault - it works this way.

closed 2020-07-23 by extrad