Napsty/check_smart #58: status line 2000GB Gigabyte AORUS M.2 2280 PCIe 4.0 x4 NVMe

michaelschefczyk, 2020-09-27:
Your plugin does work very well - thank you very much! Regarding NVMe, it did work in all my settings with Samsung devices. 

Recently, I did purchase this Gigabyte product:
2000GB Gigabyte AORUS M.2 2280 PCIe 4.0 x4 NVMe 1.3 3D-NAND TLC (GP-ASM2NE6200TTTD)

It does yield meaningful SMART outout using
- smartctl -a /dev/nvme0n1
- nvme --smart-log /dev/nvme0n1

However, the plugin does return "UNKNOWN: Drive GIGABYTE GP-ASM2NE6200TTTD S/N SN...: No health status line found".

Is there a setting available to cure this?

Regards,

Michael Schefczyk

Napsty, 2020-09-28:
Could you show the output of `smartctl -a /dev/nvme0n1` please? You can obfuscate serial numbers if you want.

michaelschefczyk, 2020-09-28:
1) smartctl -a /dev/nvme0n1 does indeed only provide general information with an empty data section

smartctl 6.6 2017-11-05 r4594 [x86_64-linux-4.19.0-11-amd64] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       GIGABYTE GP-ASM2NE6200TTTD
Serial Number:                      SN202808913818
Firmware Version:                   EGFM11.3
PCI Vendor/Subsystem ID:            0x1987
IEEE OUI Identifier:                0x6479a7
Total NVM Capacity:                 2,000,398,934,016 [2.00 TB]
Unallocated NVM Capacity:           0
Controller ID:                      1
Number of Namespaces:               1
Namespace 1 Size/Capacity:          2,000,398,934,016 [2.00 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            6479a7 39d21135fa
Local Time is:                      Mon Sep 28 20:32:08 2020 CEST
Firmware Updates (0x12):            1 Slot, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL Self_Test
Optional NVM Commands (0x005d):     Comp DS_Mngmt Wr_Zero Sav/Sel_Feat Timestmp
Maximum Data Transfer Size:         512 Pages
Warning  Comp. Temp. Threshold:     90 Celsius
Critical Comp. Temp. Threshold:     95 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     9.78W       -        -    0  0  0  0        0       0
 1 +     6.75W       -        -    1  1  1  1        0       0
 2 +     5.23W       -        -    2  2  2  2        0       0
 3 -   0.0490W       -        -    3  3  3  3     2000    2000
 4 -   0.0018W       -        -    4  4  4  4    25000   25000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         2
 1 -    4096       0         1

=== START OF SMART DATA SECTION ===
Read NVMe SMART/Health Information failed: NVMe Status 0x2002

2) smartctl -d nvme -Hi /dev/nvme0n1 is even more brief

smartctl 6.6 2017-11-05 r4594 [x86_64-linux-4.19.0-11-amd64] (local build)
Copyright (C) 2002-17, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       GIGABYTE GP-ASM2NE6200TTTD
Serial Number:                      SN202808913818
Firmware Version:                   EGFM11.3
PCI Vendor/Subsystem ID:            0x1987
IEEE OUI Identifier:                0x6479a7
Total NVM Capacity:                 2,000,398,934,016 [2.00 TB]
Unallocated NVM Capacity:           0
Controller ID:                      1
Number of Namespaces:               1
Namespace 1 Size/Capacity:          2,000,398,934,016 [2.00 TB]
Namespace 1 Formatted LBA Size:     512
Namespace 1 IEEE EUI-64:            6479a7 39d21135fa
Local Time is:                      Mon Sep 28 20:38:25 2020 CEST

=== START OF SMART DATA SECTION ===
Read NVMe SMART/Health Information failed: NVMe Status 0x2002

3) nvme --smart-log /dev/nvme0n1 does yield information one would like to monitor, however

nvme --smart-log /dev/nvme0n1
Smart Log for NVME device:nvme0n1 namespace-id:ffffffff
critical_warning                    : 0
temperature                         : 48 C
available_spare                     : 100%
available_spare_threshold           : 5%
percentage_used                     : 0%
data_units_read                     : 372944
data_units_written                  : 6006787
host_read_commands                  : 1016341
host_write_commands                 : 28611931
controller_busy_time                : 515
power_cycles                        : 27
power_on_hours                      : 38
unsafe_shutdowns                    : 22
media_errors                        : 0
num_err_log_entries                 : 37
Warning Temperature Time            : 0
Critical Composite Temperature Time : 0
Thermal Management T1 Trans Count   : 0
Thermal Management T2 Trans Count   : 0
Thermal Management T1 Total Time    : 0
Thermal Management T2 Total Time    : 0

Napsty, 2020-09-30:
It looks like this Gigabyte drive is not yet detected by `smartctl` (smartmontools). The only Gigabyte drive I can find in the drivedb is a GIGABYTE GP-* drive:
https://github.com/smartmontools/smartmontools/blob/5691bd0f65f3c2f4500457bec49869cd43b6edb4/smartmontools/drivedb.h#L689

Try it again with the currently newest smartmontools release (7.1 as of this writing) and see if the NVMe SMART information is showing up. If not, I suggest you open an issue/request in the smartmontools repo: https://github.com/smartmontools/smartmontools/issues .

Napsty, 2020-10-01:
Closing this ticket as check_smart depends on the output of smartctl. And smartctl does not detect this M2 correctly or the M2 card itself does not give correct SMART information. Anyway, following up in https://github.com/smartmontools/smartmontools/issues would be necessary.

closed 2020-10-01 by Napsty