Napsty/check_smart #74: No performance data on NVMe drive

Napsty, 2021-12-09:
```
root@bullseye:~# /usr/lib/nagios/plugins/check_smart -d /dev/nvme1n1 -i nvme
OK: Drive  UCS-SDHPCIE 800GB S/N XXX: no SMART errors detected. |=0x00 =42 =100 =10 =0 =242 =2913064 =12586 =13282120 =26 =57 =4124 =44 =0 =0
```

Performance data is there - but the keys are empty

Napsty, 2021-12-10:
Regression caused by #72

PVasileff, 2021-12-10:
Thanks. Working perfect. Tested on:

`# /usr/local/nagios/libexec $ /usr/local/nagios/libexec/smart.pl -d /dev/nvme0n1 -i nvme
`
`OK: Drive  XPG GAMMIX S11 Pro S/N 2K522L19D1WE: no SMART errors detected. |Temperature=43 Available_Spare=100 Available_Spare_Threshold=10 Percentage_Used=1 Data_Units_Read=16304228 Data_Units_Written=9256161 Host_Read_Commands=118926670 Host_Write_Commands=155221393 Controller_Busy_Time=3337 Power_Cycles=204 Power_On_Hours=3522 Unsafe_Shutdowns=21 Media_and_Data_Integrity_Errors=0 Error_Information_Log_Entries=0 Warning__Comp_Temperature_Time=0 Critical_Comp_Temperature_Time=0 Thermal_Temp_1_Transition_Count=16 Thermal_Temp_1_Total_Time=95
`

closed 2021-12-10 by Napsty