Napsty/check_smart #78: megaraid,N not work with 6.12

PVasileff, 2021-12-10:
I have configured few checks on hdd disks on my AVAGO MegaRAID SAS 9361-8i controler:

```
command[check_smart_storage_drive0]=/usr/local/nagios/libexec/smart.pl -d '/dev/sda' -i 'megaraid,23' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
command[check_smart_storage_drive1]=/usr/local/nagios/libexec/smart.pl -d '/dev/sda' -i 'megaraid,24' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
command[check_smart_storage_drive2]=/usr/local/nagios/libexec/smart.pl -d '/dev/sda' -i 'megaraid,25' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
command[check_smart_storage_drive3]=/usr/local/nagios/libexec/smart.pl -d '/dev/sda' -i 'megaraid,26' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45

command[check_smart_backups_drive0]=/usr/local/nagios/libexec/smart.pl -d '/dev/sdb' -i 'megaraid,27' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
command[check_smart_backups_drive1]=/usr/local/nagios/libexec/smart.pl -d '/dev/sdb' -i 'megaraid,28' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
command[check_smart_backups_drive2]=/usr/local/nagios/libexec/smart.pl -d '/dev/sdb' -i 'megaraid,29' -r Current_Pending_Sector,Reallocated_Sector_Ct,Program_Fail_Cnt_Total,Uncorrectable_Error_Cnt,Offline_Uncorrectable,Runtime_Bad_Block,Reported_Uncorrect,Reallocated_Event_Count,Command_Timeout,Temperature_Celsius,Airflow_Temperature_Cel -w Airflow_Temperature_Cel=45,Temperature_Celsius=45
```

```
-----------------------------------------------------------------------------------
EID:Slt DID State DG     Size Intf Med SED PI SeSz Model                   Sp Type 
-----------------------------------------------------------------------------------
21:0     24 Onln   0 1.818 TB SATA HDD N   N  512B Hitachi HUA723020ALA640 U  -    
21:1     25 Onln   0 1.818 TB SATA HDD N   N  512B HGST HUS724020ALA640    U  -    
21:2     26 Onln   0 1.818 TB SATA HDD N   N  512B HGST HUS724020ALE640    U  -    
21:3     23 Onln   0 1.818 TB SATA HDD N   N  512B HGST HUS724020ALA640    U  -    
21:4     27 Onln   1 1.818 TB SATA HDD N   N  512B Hitachi HUA723020ALA640 U  -    
21:5     29 Onln   1 1.818 TB SATA HDD N   N  512B Hitachi HUA723020ALA640 U  -    
21:6     28 Onln   1 1.818 TB SATA HDD N   N  512B Hitachi HUA723020ALA640 U  -    
-----------------------------------------------------------------------------------
```

but after update the smart.pl with latest version, when I run:

# /usr/local/nagios/libexec/smart.pl -d '/dev/sda' -i 'megaraid,24'
invalid interface megaraid,24 for /dev/sda!

With new version 6.12 on another server with nvme all works fine and performance_data work correctly but on that version of plugin I unable to check smart on my hdd drives

Napsty, 2021-12-10:
Yeah, my fault. See #76 . It's fixed in 6.12.1.

closed 2021-12-10 by Napsty

PVasileff, 2021-12-10:
I see that fixed on 6.12.1 thanks.