Napsty/check_smart #95: No output after pipe

goudeuk2000, 2023-10-06:
Hello

First of all thank you for your time in writing this very useful plugin.

When I try it ( check_smart -d /dev/sda -i auto) I only get some of the output instead of the complete result as shown in the examples.

I only get this. Nothing is shown after the pipe 

**OK: Drive  DELL MD32xx S/N 3C3005A: no SMART errors detected. |**

Can you please let me know if I am doing something wrong? Is there something I need to change to make it work?

Thank you

Napsty, 2023-10-20:
Hi there. How exactly did you execute the plugin? Can you please show the full command.

goudeuk2000, 2023-10-20:
Hi

I run the command on the local machine which also has nagios on it. From inside the directory: /usr/local/nagios/libexec ,
I execute: 
**./check_smart -d /dev/sdb -i auto**
 
Thank you

Napsty, 2023-10-20:
I assume you are using a DELL PERC RAID controller? This is similar to MegaRAID controllers. Then you have to use a different syntax. See examples on https://www.claudiokuenzler.com/monitoring-plugins/check_smart.php#Usage .

I suggest to first run `sudo smartctl --scan` to identify the physical drives.

goudeuk2000, 2023-10-20:
Hi Napsty,

**smartctl --scan** = very helpful. 
Thank you very much. Yes, it is megaraid. I get all the info now. 

All the best.

closed 2023-10-20 by Napsty