Napsty/check_smart #96: Kingston ssd wearout not detected

sthibaul, 2024-02-17:
Hello,

Some SSD disks unexpectedly failed, and we have noticed this difference between good devices and failing devices:

good:

    169 Remaining_Lifetime_Perc 0x0000   093   093   000    Old_age   Offline      -       93

bad:

    169 Remaining_Lifetime_Perc 0x0000   000   000   000    Old_age   Offline      -       0

So this probably needs to be added to a list of values to check.

Samuel

Napsty, 2024-02-17:
Similarity to #86 . It seems that attribute 169 (Remaining_Lifetime_Perc) is a decreasing counter and needs to be handled differently than other attributes (including 202 Percent_Lifetime_Remain).

sthibaul, 2024-02-20:
They also have

    231 SSD_Life_Left           0x0000   000   000   000    Old_age   Offline      -       0

Napsty, 2024-02-21:
> 231 SSD_Life_Left

Is this a decreasing or increasing counter?

sthibaul, 2024-02-21:
It's decreasing as well, that was the output from the old disk