Napsty/check_smart #98: Incorrect output with sat+megaraid,[0-3]

MegaS0ra, 2024-04-22:
Hello,

I can get correct results with this command : 
`./check_smart.pl -g /dev/sda -i sat+megaraid,0`
`OK: [sat+megaraid,0] - Device is clean|`

But not with this one : 
`./check_smart.pl -g /dev/sda -i 'sat+megaraid,[0-3]'`
`UNKNOWN: [megaraid,0] - No health status line found[megaraid,0] - [megaraid,0] -  --- [megaraid,1] - No health status line found[megaraid,1] - [megaraid,1] -  --- [megaraid,2] - No health status line found[megaraid,2] - [megaraid,2] -  --- [megaraid,3] - No health status line found[megaraid,3] - [megaraid,3] - |`

It seems with the second command, "sat+" is not kept.
Would the command be different in this case, any character to escape maybe ?

Thanks

Napsty, 2024-04-22:
Hi. Did you try it with `./check_smart.pl -g /dev/sda -i "megaraid,[0-3]"` ?

MegaS0ra, 2024-04-22:
Yes :
`./check_smart.pl -g /dev/sda -i "megaraid,[0-3]"`
`UNKNOWN: [megaraid,0] - No health status line found[megaraid,0] - [megaraid,0] -  --- [megaraid,1] - No health status line found[megaraid,1] - [megaraid,1] -  --- [megaraid,2] - No health status line found[megaraid,2] - [megaraid,2] -  --- [megaraid,3] - No health status line found[megaraid,3] - [megaraid,3] - |`

I suppose it does not work because when using smartctl, I have to use sat+megaraid : 
`smartctl -a /dev/sda -d sat+megaraid,0`

But I just realized this is related to the version of smartctl used :
On servers with smartctl 5.43, I have to use "sat+megaraid", whereas on servers with version 6 or 7, I can use only "megaraid" without any issue.
Upgrading the version of smartctl solved my issue.

Sorry for bothering, and thanks for checking on this !

closed 2024-04-22 by MegaS0ra

Napsty, 2024-04-22:
Whoa, smartctl 5.43 is OOOLD ;-) 
Great that it works now with the newer version, thanks for the quick feedback.